pioneers/runtime #20: [ORGANIZATION] look into reorganizing files into larger directories

benliao1, 2020-06-24:
Potential organizations to consider:
```
/c-runtime/runtime/src
                  /include
                  /bin
                  /test
                  /demos
                  /Makefile
         /lowcar/lib
         /flash.sh
         /Makefile
```
Or 
```
/c-runtime/production/executor
                     /shm_wrapper
                     /dev_handler
                     /net_handler
                     /logger
                     /runtime_util
                     /Makefile
         /test/executor
              /dev_handler
              /net_handler
              /logger
              /Makefile
         /lowcar/lib
                /flash.sh
                /Makefile
```

AVSurfer123, 2020-07-15:
I actually prefer the current organization over this one

benliao1, 2020-07-19:
Yeah I agree actually, looking back at this. We should bring it up at the next meeting and then decide whether to close this and #7

closed 2020-07-22 by benliao1