pioneers/runtime #39: [EXECUTOR] Parse student code to determine subscription bitmap

benliao1, 2020-07-08:
Executor can parse student code and determine which parameters of which devices are used in student code, and set the subscription bitmap for the executor appropriately. This ensures that the device handler will fetch these values from the indicated devices and supply them to the executor / student code.

If the student code indicates for a device that does not exist and the code is running in `TELEOP` or `AUTO` mode, executor should issue a warning to the student code that the indicated device is not attached, but not crash the code. Rather, it should keep running, and return some sort of default value for those values.

AVSurfer123, 2020-07-09:
This seems way harder to implement than the benefit gained by it. It's even harder to implement because we are using C and the string parsing sucks.

Also I just checked the SHM code and it seems that the executor currently has no way to determine if the UID doesn't exist in SHM since the SHM don't return an error status. So executor would just be giving garbage value on read.

However, the executor should definitely crash if it can't find the device. It would be very bad to catch that exception as it's the only way for the student to know that their devices aren't connected/working. We should only catch errors that are caused by us.

benliao1, 2020-07-10:
It's actually a really key part if we want to reduce the amount of computation that the device handler does by a significant factor. Take, for example, the KoalaBear, which may end up having 25+ parameters. Without the subscription bitmap, the device handler runtime would (be default) subscribe to all of these parameters. This means that the dev handler would need to decode all 25 of these parameters 10, 20 or 30 times a second (depending on how quickly the Arduino can run, and our requested subscription). But the students are almost certainly not going to be using all 25 of these parameters (perhaps they only use 2--the duty cycles on both connected motors), in which case the device handler can support the student code while doing <5% of the original work it was doing. 

I do agree that we need to figure out exactly what the load on the device handler is before we implement it though, and that this part of it in particular is going to be very difficult--which is why it'll be in v1.1 if it's implemented at all.

Regarding what happens when a student specifies an unregistered device, right now it simply crashes the student code without returning the robot mode to `IDLE` right?

AVSurfer123, 2020-07-10:
No like I said the student API has no way of knowing whether the device actually exists since SHM doesn't give any return status. It just silently returns after it catches the exception. So device_write does nothing if the device doesn't exist, and device read will return garbage values from what it malloced. I should add an issue to fix that, but its pretty simple. You just need to change the return type from void to int.

AVSurfer123, 2020-07-20:
There are still some bugs with code parsing, the major one being that it only can handle when `Robot.set_value` and `Robot.get_value` have their first two inputs either be literals or be global variables. If they are local variables, it will fail to correctly send a subscription.

closed 2020-07-21 by AVSurfer123