pioneers/runtime #55: [LOGGER] Logging with PYTHON level doesn't work

AVSurfer123, 2020-07-17:
Python errors aren't getting sent over the network properly, with many lines being cut out.

benliao1, 2020-07-17:
We found an issue similar to this (except it was double printing) in `dev_handler`
Perhaps try copy-pasting the `logger.c` and `logger.h` files in the newest commit on `dev_handler` branch and see if that fixes this problem (we added a mutex to make sure logs are being emitted one at a time, there was no synchronization check before)

benliao1, 2020-07-17:
Fixed, I think! With edit above

AVSurfer123, 2020-07-19:
This was caused by adding an extra byte for the null character when the logger sends data to the log FIFO. We don't want to send the null character due to how `fgets` works.

closed 2020-07-19 by AVSurfer123