pioneers/runtime #68: [DOCS] Update documentation

benliao1, 2020-07-24:
Right before we put a `v1.0` on a commit, we should create a branch called like `update_docs` or something and all of us just go in and do the following:

- Add function documentation to all of our headers and make sure it's up to date
- Transfer parts of the various `README`s over to the Wiki
- Fill out the Wiki pages (and add more if needed)
- Read through all of our code one more time and make sure all comments are up-to-date, adding / deleting more as necessary

That branch should be the final PR before `v1.0`!

benliao1, 2020-08-05:
Behavior Google Doc: https://docs.google.com/document/d/17WZzOFiMEmu7n7rX26YW1kB-g1vKGecYK6eZqyUovOc/edit?usp=sharing (use PiE email)

AVSurfer123, 2020-08-05:
Standard tests should point to issue #71