pioneers/runtime #76: [TEST] Use gcov and gprof for code analysis / optimization

benliao1, 2020-07-28:
`gcov` is a code coverage tool, and `gprof` is a code timing analysis tool. This will help us understand where our code is spending most of its time executing (`gprof`) and will also help us figure out how much of our code isn't being tested (`gcov`). This will be really useful in the future, but right now, we don't have enough tests to justify using this, so it should definitely be a 1.1 feature.

Thanks to @charleshong3 for the suggestion!

AVSurfer123, 2020-08-07:
@charleshong3 Can you help do this, or give tips on how/whether to do this?

benliao1, 2020-08-07:
I looked into this for a bit and I think basically you compile the code with some extra flags, and then after you're done running your tests, you'll have some extra files you can analyze using gcov and gprof. From that, you can determine code coverage. But I didn't experiment with it at all or test it either; it's definitely a low priority.

charleshong3, 2020-08-16:
Not sure about gprof, but with gcov you add the `--coverage` flag when compiling with `gcc` or `clang`. Then you do `gcov <file1>.c <file2>.c ...` and it tells you your code coverage along with generating a report.

You can augment this with [lcov](http://ltp.sourceforge.net/coverage/lcov.php), a visualizer, or fancy tools like [codecov](https://docs.codecov.io/docs/pull-request-comments) which seem really cool but I don't actually have any experience with.

benliao1, 2022-07-21:
Will mark as `wontfix` because while it would be nice, I don't think we will ever realistically implement this.

closed 2022-07-21 by benliao1