pioneers/runtime #82: [MISC] Investigate EOF on reading from log pipe on net_handler

benliao1, 2020-07-29:
This is a strange bug that was observed when running `tc_68_2`. For some reason, the TCP connection was reading EOF from the pipe constantly during shutdown of the processes at the conclusion of the test, generating a `Suppressing output; too many messages...` on `net_handler_client`. The only way for FIFO pipes to generate EOFs is if all write ends of the pipes are closed and the read end is still attempting to read from the pipe. But this shouldn't happen, because the TCP thread should be canceled (and therefore the read end of the pipe closed) before the net handler process quits (when the last write end should close). For now, an ad-hoc fix was implemented by sticking an EOF check in `send_log_msg` in `tcp_conn.c`, but the source of the problem is still unknown.

AVSurfer123, 2020-12-19:
Stale bug, probably has been solved by now lol.

AVSurfer123, 2021-01-17:
Nvm probably still an issue

levincent06, 2021-01-18:
# Steps to Reproduce
Commit `0e45741` (Just merged in #154)
1. Save `studentcode.py` as below
2. Open SHM UI
3. Open Executor CLI (Which should automatically use `studentcode.py`)
4. Open Net Handler CLI and send `TELEOP`
5. `send_log_msg()` will be called many times with an `EOF`. To see this, add a print statement [before calling the function](https://github.com/pioneers/runtime/blob/0e45741d9635404afeee05d0e394d0bb7bd939cd/net_handler/tcp_conn.c#L299-L302).

Note that Executor will try to access shared memory for device `62_99999` which is NOT connected. This is intentional and an expected `DeviceError` should be raised.
## Student Code
```
MOTOR = "62_99999"
def autonomous_setup():
    print('Running autonomous action ...')
def autonomous_main():
    pass
def teleop_setup():
    Robot.get_value(MOTOR, "FLIP_FLOP");
    pass
def teleop_main():
    pass 
```