pioneers/runtime #83: [TEST] Make test framework able to stop and restart processes

benliao1, 2020-07-30:
For some reason, the test framework hangs when starting and stopping the executor, not really sure why. We need to make sure that the test framework is able to stop and start all Runtime processes smoothly, so we can test the behavior of Runtime when processes crash and come back up.

benliao1, 2020-07-31:
To reproduce this bug, add the following lines between the end of the `poke` section and before the `end test` section of `tc_68_2`:
```
stop_executor();
start_executor("executor_sanity");
```
Running the test, you should see the errors (or something very similar):
```
kill: child process does not exist
wait: child process does not exist
```
and then it hangs until you press `Ctrl-C`, and then it cleans up properly.

My original theory was that the executor client was installing the `SIGINT` handler on the executor twice, but when I added a global flag in the executor client to make sure that the `SIGINT` handler was only installed once, it still failed, so I'm not sure what's going on.

AVSurfer123, 2020-08-29:
This was fixed when I removed the duplicate/incorrect signal handlers.

closed 2020-08-29 by AVSurfer123