pioneers/runtime #119: [MISC] Investigate SoundDevice playing a continuous note

levincent06, 2020-09-20:
We want to be able to hold a button down and a mapped note will play for as long as that button is pressed (like a normal piano). 

As of now, `sound.py` has a global variable `duration` that sets how long to play a note when it reads it over TCP. Furthermore, the student code needs `time.sleep()`s after writing a pitch or else there is some latency with button presses on the executor(?) side.

This issue needs to be further investigated... what is the "sweet-spot" for the `duration` time? Are the `sleep()`s in the student code necessary? Do they need to match the `duration` variable in `sound.py`?

Previous experiments to get this working made notes play without stopping or the notes are "buzzy" (this is likely due to how the sound wave is implemented in `sound.py`).

benliao1, 2021-09-22:
We will probably just remove the SoundDevice :\

AVSurfer123, 2021-09-25:
Noooooo we can't remove it. I will make this work at some point...

levincent06, 2022-07-21:
sure lmk when

closed 2022-07-21 by levincent06