pioneers/runtime #135: Create repo for GoPro Video Server

AVSurfer123, 2020-12-19:
Also some nice resources I found on programmatically accessing its video feed.

https://github.com/KonradIT/goprowifihack
https://github.com/KonradIT/gopro-py-api
https://www.helmutsteiner.com/2020/04/14/how-to-use-your-gopro-as-a-webcam-with-obs-studio-no-capture-card-required/
https://goprohero.readthedocs.io/en/latest/

mberenshteyn, 2020-12-19:
Wrote some notes about the connection/streaming setup process in https://docs.google.com/document/d/1abH0NvghXWey8Pa6wZ5qbiSzuUx2bXQ-6yXjN-MzG-Y/edit?usp=sharing

benliao1, 2021-03-20:
The server was handled with a single Python script (and a whole bunch of setup) but probably don't need a repo for that one script. We will put that Python script into a Github wiki page instead, since it doesn't really belong in the Runtime repo proper.

closed 2021-03-20 by benliao1