pioneers/runtime #136: [FATAL] Net Handler "Failed to malloc" on send_log_msg()

levincent06, 2020-12-20:
Net Handler CLI fails to malloc after being idle for ~3 minutes. After this happens, sending another run mode command will exit the net handler cli without warning. Attempting to restart net handler cli won't work (`Client is neither Dawn nor Shepherd`) and any attempts to switch run mode won't register in shared memory. The fix for now is to simply restart the Raspberry Pi when this happens.

Steps to reproduce(?):
1) Run all four components of runtime with a connected KoalaBear
2) Switch between IDLE/AUTO multiple times
3) Wait a few minutes

Might be able to reproduce by having just shared memory and net handler. 
May or may not be linked to 
- switching run modes repeatedly
- Received bad message (type 6) from KoalaBear when connected
- Incorrect hex uid printed in dev handler (Expected `0x123456..` but showing `0x23456...`)

levincent06, 2020-12-26:
- Fixed "Received bad message (type 6)" in Pull Request #138 
- Fixed net handler indexing bug in Pull Requested #137

levincent06, 2020-12-26:
- Investigating incorrect hex uid in issue #139 (Which is probably not related to this issue)

benliao1, 2021-01-02:
Added some debugging logs and prints messages in `tcp_conn.c` and `logger.c` to try and catch this error when it occurs. Only happens on Vincent's raspi and Ashwin's docker-on-raspi.

levincent06, 2021-01-14:
Update! This was reproducible on Daniel's Pi as well. Steps to reproduce (consistently!)
Write student code that refers to connected devices in the global variables. i.e. This code is valid, and should run as expected
This is sufficient:
```
MOTOR = '6_<valid uid>'
def autonomous_setup():
    pass
def autonomous_main():
    pass
def teleop_setup():
    pass
def teleop_main():
    pass
```
## Steps to Reproduce
### Standard Dawn/Runtime Setup
1. Have Dawn open. Upload a valid student code file if it's not already on the Pi.
2. Start up Runtime manually without `systemd`
   a. SHM Dashboard
   b. Have KoalaBear connected with uid `<valid uid>` in student code, then open Dev Handler
   c. Net Handler (Dawn should autoconnect)
   d. Executor CLI (Should automatically use our valid student code)

### Standard Run Mode Switching
3. Dawn: Send `TELEOP`
4. Dawn: Send `IDLE`

### Upload invalid student code
5. Dawn: Change the `MOTOR` uid to something invalid. Changing one digit should be fine.
6. Dawn: Upload the invalid student code
7. Executor CLI: Refresh the student code being used
    i. `student code`, followed by `studentcode`
  
### Execute invalid student code
8. Dawn: Send `TELEOP` -- Executor CLI should throw a `DeviceError` because no such device with the specified uid is connected
9. Dawn: Send `IDLE`

### Observe error
10. Wait 3 to 4 minutes.
11. See Fail to malloc payload error

We'll continue to investigate this issue and see if we can reproduce it with as few steps as possible.

levincent06, 2021-01-18:
The problem was due to `send_log_msg()` being called over 20 times per second and returning due to an `EOF` on the log FIFO. The fix is to properly `free()` any allocated memory before returning. We also require that there is no `EOF` before calling this function. Issue #82 has been re-opened to investigate this EOF.

PR #173 (2021-01-23): [NET_HANDLER] Fix memory leak in send_log_msg()
- Free all allocated memory before returning
- Call function iff FIFO does not have an EOF

Resolves #136 -- We were not properly `free()`ing memory before returning

Finding this bug made it clear that `tcp_conn.c` is still being spammed with `EOF`, even though not all write-ends are closed.
As a result, we re-opened #82
closed 2021-01-23 by levincent06