pioneers/runtime #142: [SYSTEMD] Add script to start ngrok and post tunnel URL

levincent06, 2021-01-03:
For the remote Spring 2021 competition, Dawn and Shepherd will be connected to Runtime via a ngrok tunnel instead of an IP address on a local network. Normally, students can check their robots' IP address by reading it off a label on its router, which is not possible for the remote competition. Now the students need to know the robots' ngrok tunnel URL, which may be different every time ngrok is restarted. 

Runtime's systemd should start ngrok (in addition to Runtime itself), get the tunnel URL, then post it online so that PIE staff can send it to the students. The goal is that PIE staff can open something easily accessible (ex: Google Doc) instead of ssh-ing into the Rasp Pi to find it.

AVSurfer123, 2021-01-06:
After doing a bit of research, it seems that `ngrok` can only forward the TCP connection, and not the UDP connection. So we would have to write or use another program to either tunnel the UDP connection, or send the UDP datagram as a TCP packet.

levincent06, 2021-01-07:
Is this going to cause problems because we will lose the advantage of speed and ignoring late packets? It may be an issue if Runtime responds to a delayed gamepad input. This might be a larger software discussion @benliao1

levincent06, 2021-01-09:
Update: We've decided to not get the tunnel URL and post it online. This is because the ngrok website has a dashboard that shows it already. (We still want a script that starts ngrok on startup.)

closed 2021-02-06 by DanielMolina24