pioneers/runtime #161: [MISC] Create Wiki page about Student API

benliao1, 2021-01-11:
Pimulator, leadership, Dawn, and piewebserver have all requested that we create some sort of master documentation for the student API. We can put it in a Wiki page with a "Last Updated on: <date>" at the top which can be a source of truth for all other implementations or reference documentation put out by PiE regarding the StudentAPI.

AVSurfer123, 2021-01-12:
There's already a wiki page on it. But this is more of documentation, which you can probably autogenerate/copy from the doc strings I had.

benliao1, 2021-01-12:
Hmm yeah I did see the wiki page, at least we should add the stuff about the Keyboard and format it and stuff. I don't personally think we need to go through all the trouble of producing documentation from our code but idk.. maybe it's easier than I think it will be :P

closed 2021-01-15 by benliao1