pioneers/runtime #236: [DEV_HANDLER][LOWCAR] Add Fin message

benliao1, 2021-09-22:
When one side of the connection between the Raspberry Pi and the device times out, the side that timed out should send a FIN on the wire to the receiver to potentially let them know that they have been timed out and should therefore reset the connection. Currently, this is implemented as a timeout on both sides (i.e. when one side times out, it will wait for a predetermined amount of time to give the other end time to timeout and reset), but this is inconsistent, as we found in PR #234.

levincent06, 2021-09-23:
We closed this by accident

levincent06, 2021-10-21:
Dev Handler's exit handler should send a `FIN` to any currently connected devices. This should prevent connected motors from spazzing out whenever we `pkill -6 dev_handler`

commit d9bd80b0 (2021-09-22): [DEV_HANDLER] Increase TIMEOUT to 2s. Temp fix #236
closed 2023-06-18 by BrandonWong14