pioneers/runtime #240: [NET_HANDLER] Accept WebSocket connections in addition to normal TCP sockets

levincent06, 2021-09-23:
Dawn will be implementing a browser-based application, which will communicate with WebSockets instead of TCP sockets. We need to support browser-Dawn in addition to desktop app Dawn.

Here is an explanation for the difference between WebSockets and normal sockets: https://stackoverflow.com/q/4973622 (tl;dr WebSockets are built on top of TCP sockets)
Here is a small library that we can use: https://theldus.github.io/wsServer/

benliao1, 2021-09-24:
We may be rehashing the implementation of this. We will probably use the Python websocket library and start a new process, like websocket_server.py, which connects to `localhost:RASPI_TCP_PORT` on one end and a new websocket server on the other end and just ferries messages between the two. We might even be able to support both web browser Dawn and electron app Dawn simultaneously with this design.

AVSurfer123, 2021-09-25:
We will be using this library for Python https://websockets.readthedocs.io/en/stable/